grbl/grbl-sim #7: socat without sudo?

winder, 2015-05-29:
On my LinuxMint machine I needed to run socat as sudo, at which point the /dev/ttyFAKE device is owned by root.root instead of root.dialout. At this point my GUI application can connect to /dev/ttyFAKE just fine when I run it as root, but I'd rather not have to run my IDE/debugger as root if I can find a way to avoid that.

ashelly, 2015-05-29:
I avoid running socat as sudo by setting the devicename to /tmp/ttyFAKE instead.  see [simport.sh](https://github.com/grbl/grbl-sim/blob/master/simport.sh).    This works for my python client.

Another option:  run socat as sudo, then `sudo chmod a+rw /dev/ttyFAKE`.  After doing this, I 
I got gtkterm to connect to `/dev/ttyFAKE`  when running under my own user ID.

winder, 2015-05-29:
Thanks, chmod works. I was trying chown before and the owner wasn't being changed.

simport.sh isn't working though, it gives me different error when I run it, I tried a few different directories besides /tmp that my user should have write access to:
```
fopen: Permission denied
2015/05/29 13:53:44 socat[9784] W waitpid(): child 9785 exited with status 255
```

winder, 2015-06-08:
I worked around this with pull request #8

closed 2015-06-08 by winder