grbl/grbl-sim #10: cannot link grbl-sim to a com port on Windows

simonmtl, 2015-09-04:
After compiling the code on windows 7, I installed "Virtual Com Port Driver" but I realized I have no handle to connect the executable with the com port. I tried a simple redirect > but that's unidirectional so it's a Nogo.

Is there a way to do that on windows without implementing it the code?

winder, 2015-09-04:
I've used this program in the past http://com0com.sourceforge.net/

simonmtl, 2015-09-15:
Thanks, I'm probably missing something.  I don't know how to bind grbl_sim.exe to the virtual com port. The grbl_sim.exe doesn't appear to know anything about com ports, virtual or not. 

The directions you sent me mention a scriptable terminal emulator (couldn't find it online), does that mean that grbl_sim could act as the script?

winder, 2015-09-15:
Sorry, I haven't used it with grbl_sim.exe since my primary workstation is a linux machine. It looked like it might be a possible solution so I gave a link.