grbl/grbl-sim #9: How I compiled grbl-sim on windows

simonmtl, 2015-09-04:
I had to make the following changes on the makefile to be able to compile with ming32-make on win7:

-change relative paths from ../ to ..\ 
-change rm -f command to del 
-$(CC) was not recognised, I changed the line to gcc

To avoid messing with the makefile, I tried to compile through Cygwin but file conio.h was not found (it's a windows file), so I changed it to curses.h but then some function calls were not identified, so I gave up on this approach although it's probably simple to correct/modify.

ashelly, 2015-09-14:
Thanks for the feedback.  I will address when I get some bandwidth.

thrustin, 2018-08-21:
I am having trouble trying to compile on windows. Is it possible to upload your makefile?

hansmbakker, 2019-02-02:
Please see #19 and #20