grbl/grbl-sim #11: Windows version is not 'stdin' redirect friendly

alexian79, 2015-11-02:
Hi guys, 
with original code I was not able to use < input redirection on Windows, to read g-code from file. I think root cause is using _kbhit() and getch().

Did quick fix that works for me:

From 05fd7ab068552fc7955d9d50132efa727da3818f Mon Sep 17 00:00:00 2001
From: Oleksiy Bondarenko <alexeibondarenko@cashcode.com>
Date: Mon, 2 Nov 2015 12:51:01 -0500
Subject: [PATCH] Made windows version stdin redirect friendly

---
 Makefile           |  4 ++--
 platform_WINDOWS.c | 32 +++++++++++++++++++++++++++-----
 sim.bat            |  4 +++-
 3 files changed, 32 insertions(+), 8 deletions(-)

diff --git a/Makefile b/Makefile
index a6945bd..298123f 100644
--- a/Makefile
+++ b/Makefile
@@ -16,9 +16,9 @@
 #  You should have received a copy of the GNU General Public License
 #  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
 
-# PLATFORM   = WINDOWS
+PLATFORM   = WINDOWS
 # PLATFORM   = OSX
-PLATFORM   = LINUX
+# PLATFORM   = LINUX
 
 #The original grbl code, except those files overriden by sim
 GRBL_BASE_OBJECTS =   ../protocol.o ../planner.o ../settings.o ../print.o ../nuts_bolts.o  ../stepper.o ../gcode.o ../spindle_control.o ../motion_control.o ../limits.o ../coolant_control.o ../probe.o ../system.o 
diff --git a/platform_WINDOWS.c b/platform_WINDOWS.c
index 7c4a3c3..71b8224 100644
--- a/platform_WINDOWS.c
+++ b/platform_WINDOWS.c
@@ -31,12 +31,32 @@
 
 double ns_per_perfcount;
 
+volatile struct {
+    //CRITICAL_SECTION m_lock;
+    bool isEmpty;
+    int content;
+} stdin_q;
+
+
+PLAT_THREAD_FUNC(stdin_poller, exit)
+{
+    while (true) {
+        if (stdin_q.isEmpty) {
+            stdin_q.content = getchar();
+            stdin_q.isEmpty = false;
+        }
+        Sleep(1);
+    }
+    return NULL;
+}
+
 //any platform-specific setup that must be done before sim starts here
 void platform_init()
 {
   __int64 counts_per_sec;
   QueryPerformanceFrequency((LARGE_INTEGER*)&counts_per_sec);
   ns_per_perfcount = (float)NS_PER_SEC / counts_per_sec;
+  platform_start_thread(stdin_poller);
 }
 
 //cleanup int here;
@@ -62,7 +82,6 @@ void platform_sleep(long  microsec)
 {
   Sleep(microsec/MICRO_PER_MILLI);
 }
-
   
 //create a thread
 plat_thread_t* platform_start_thread(plat_threadfunc_t threadfunc) {
@@ -89,8 +108,11 @@ void platform_kill_thread(plat_thread_t* th){
 
 //return char if one available.
 uint8_t platform_poll_stdin() {
-  if (_kbhit()) {
-     return getch();
-  }
-  return 0;
+    if (!stdin_q.isEmpty) {
+        volatile int tmp = stdin_q.content;
+        putchar(tmp);
+        stdin_q.isEmpty = true;
+        return tmp;
+    }
+    return 0;
 }
diff --git a/sim.bat b/sim.bat
index ea3b05e..8dcd46b 100755
--- a/sim.bat
+++ b/sim.bat
@@ -1 +1,3 @@
-./grbl_sim.exe -t 1 0.01 <HelloWorld.nc >HelloWorld.dat 2> HelloWorldSteps.dat
+@rem grbl_sim -h
+@rem gdb --args 
+grbl_sim.exe -r 1 -t 1 -s HelloWorldSteps.dat 0.01 0<test.nc 1>out.txt
-- 
1.9.5.msysgit.1

13hakta, 2016-05-03:
Please create a pull request to this repository. And if not, please share patch as non-formatted text.

alexian79, 2016-05-12:
Will look for this change this evening.

alexian79, 2016-05-12:
change extension to .patch

PS. There was another blocker issue for me that simulator deadlocks randomly during path simulation.

[0001-Made-windows-version-stdin-redirect-friendly.patch.txt](https://github.com/grbl/grbl-sim/files/262015/0001-Made-windows-version-stdin-redirect-friendly.patch.txt)