grbl/grbl-sim #13: Grbl v1.1

chamnit, 2017-01-09:
@ashelly @jgeisler0303 : Do either of you guys have a bit of time update Grbl simulator to compile with Grbl v1.1? I'd like to use this for development testing. I don't quite understand the include voodoo you guys did. There are a lot of include errors when I tried to compile v1.1 and v0.9j. If it's something on my end, what are the system requirements? I can use vagrant to setup a quick dev virtual machine.

misan, 2017-04-22:
1.1f won't compile for OSX
`grbl_interface.c:84:6: error: use of undeclared identifier 'TCCRA_REGISTER'
  if(TCCRA_REGISTER >= 127) ocr = OCR_REGISTER;
     ^

grbl_interface.c:84:35: error: use of undeclared identifier 'OCR_REGISTER'
  if(TCCRA_REGISTER >= 127) ocr = OCR_REGISTER;
                                  ^

grbl_interface.c:94:79: error: no member named 'position' in 'system_t'
     fprintf(args.step_out_file, "%12.5f %d, %d, %d, %d\n", sim.sim_time, sys.position[X_AXIS], sys.position[Y_AXIS], sys.position[Z_AX...
                                                                          ~~~ ^

grbl_interface.c:103:79: error: no member named 'position' in 'system_t'
     fprintf(args.step_out_file, "%12.5f %d, %d, %d, %d\n", sim.sim_time, sys.position[X_AXIS], sys.position[Y_AXIS], sys.position[Z_AX...
                                                                          ~~~ ^
4 errors generated.
make: *** [grbl_interface.o] Error 1
`

chamnit, 2017-04-22:
Grbl sim is deprecated and will be replaced very soon.

closed 2017-04-22 by chamnit

misan, 2017-04-22:
@chamnit I've got it working with a few changes:
1. replacing sys.positions by sys_positions
2. disabling variable speed spindle control in config.h
3. adding jog.o to Makefile GRBL_BASE_OBJECTS

grbl_sim worked but gvalidate failed to link. Anyway, I will stay tuned for the replacement.

Thanks.

dastultz, 2017-11-19:
Has this been "replaced very soon" yet? I am trying to compile for OSX and even with the above suggestions I get:

```
Undefined symbols for architecture x86_64:
  "_ui_init", referenced from:
      _avr_main in main.o
  "_ui_service", referenced from:
      _protocol_main_loop in protocol.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [main] Error 1
```

hansmbakker, 2019-02-02:
@chamnit has Grbl Sim been replaced by something else or are there still plans for that?