grbl/grbl-sim #16: compiling grbl-sim with cygwin

thrustin, 2018-08-22:
During compiling with "make new" command I keep getting the following error.

./config.h:28:10: fatal error: ../system.h: No such file or directory
 #include "../system.h"

as well as for
#include "../planner.h"
#include "../settings.h"

Are there missing files? 

Thanks

MitchBradley, 2018-09-06:
grbl-sim must be installed as a subdirectory of the base grbl code, specifically underneath grbl/grbl

Here is how I did it:

```
git clone https://github.com/gnea/grbl.git
cd grbl/grbl
git clone https://github.com/grbl/grbl-sim.git
```

Note that grbl comes from the "gnea" user (the grbl 1.1 series), while grbl-sim comes from "grbl" (the grbl 0.x series).

After doing that, there are still problems, namely the missing file "system_declares.c".

MitchBradley, 2018-09-06:
The missing system_declares.c has now been corrected via a pull request by @fsantini 

https://github.com/grbl/grbl-sim/pull/17

chamnit, 2018-09-06:
Thanks for the confirmation

closed 2018-09-06 by chamnit