grbl/grbl-sim #18: ./config.h:28:10: fatal error: ../system.h: No such file or directory

Harvie, 2018-12-14:
```
$ LC_ALL=C make
cc -Wall -g -O3 -DF_CPU=16000000  -include config.h -I. -DPLAT_LINUX -c grbl_interface.c -o grbl_interface.o
In file included from <command-line>:
./config.h:28:10: fatal error: ../system.h: No such file or directory
 #include "../system.h"
          ^~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:64: grbl_interface.o] Error 1

```

hansmbakker, 2019-02-02:
I have the same issue when I try setting up Azure Pipelines

goeland86, 2024-03-02:
Marking the issue as closed isn't really fixing it... Just ran into the issue, and the instructions on building grbl-sim don't indicate that it depends on the original grbl code. Leaving the issue open until the readme incorporates a mention of that would be helpful.