grbl/grbl-sim #19: Is there any release available?

hansmbakker, 2019-01-29:
Is there anybody who has a working build who could add the binaries as a release?

It would really lower the barrier of using this if we could avoid having to set up a build environment :)

hansmbakker, 2019-02-02:
This can be closed when #20 is merged and set up.

I created binaries using Azure Pipelines, the result is [drop.zip](https://github.com/grbl/grbl-sim/files/2824869/drop.zip).

closed 2019-02-02 by hansmbakker

EdwinMarteUSU, 2021-03-20:
Hi, I downloaded the Binaries releases for windows. How do I specify the port to which the tool will connect to?

hansmbakker, 2021-03-21:
I have no idea, it was several years ago that I made the binaries so that others wouldn't need to go through the hassle of compiling etc.

However you're the first who showed interest in them and the repository seems abandoned unfortunately.

Since I'm not using this tool anymore I can't help you with it. Hopefully somebody else can help you.