grbl/grbl-sim #25: How to use SIM?

haoshuiwuxiang, 2020-09-06:
Hello, I successfully compiled sim. But do not know how to use it to achieve visual debugging, such as real-time display of grbl buffer block running state, or visualization speed?

flynneva, 2020-09-11:
just run the `simport.sh` script provided. you need to change the beginning of it to just `-dd` though in order to get it to work. this runs the `grbl_sim.exe` program that should be generated once you compile.

haoshuiwuxiang, 2020-10-23:
“Each of the AVR functions are replaced with dummy functions, like the stepper ISR. These could be written to do whatever you need. For example, output simulated step pulses over time and examine its performance.”
I have successfully compiled and run grbl-sim, but I actually want to use a computer to display the ISR step pulse waveform, etc., so as to realize online virtual debugging and running away from analysis. Do you know what I should do or have any suggestions? Thank you